Jean-MarcHarvengt/MCUME #1: MCUME_esp32  Compilation error

smutek110, 2021-05-20:
I can't compile. What version of esp-idf are you using during compilation?

Jean-MarcHarvengt, 2021-05-22:
It was a long time ago...
I recall I needed a specific version
I found the zip of esp-idf-v3.2 and 3.3beta3 but I think it is the 3.2

Sorry. Always hard to find back.
If you know how to find it back from the installed dir tell me. I still have it.
Also hard to find ESP32 with same pinout...

smutek110, 2021-05-25:
I have tried different versions of esp-idf. i eventually compiled with esp-idf3.2.5 and old gcc5.2, but there were bugs regarding undeclared variables. I'll try again with the esp-idf you talked about. Ultimately, if I failed, I would very much ask you to compile and provide me with the binary for flashing esp download tool. I really care about getting my son involved in electronics.

smutek110, 2021-05-25:
podczas kompilacji niektóre biblioteki masz nazwane w kodzie z małej litery a w folderze main jest nazwany z dużej litery. co powoduje błąd brak pliku. było tak w kilku emulatorach. definitywnie i tak po kompilacji nie działa .z monitora widze że ma kłopoty z tablicą partycji

Jean-MarcHarvengt, 2021-05-26:
Toolchain path: /Volumes/Macintosh MainData/_Documents/Arduino/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0

I now have an issue to compile the boot loader of the custom loader app (espboot), rest I can at leas compile but I did not try to flash and run

smutek110, 2021-05-27:
I managed to compile the applications and each one works separately, but the loader (application selector) after compilation does not work properly, it loads the last esp5200 applications automatically at startup. in esp64 and esp81 you have to rename the files and move some to the appropriate folders. It would be good to compile espboot correctly so that the application selector works.

smutek110, 2021-05-27:
I like it very much . Good job

smutek110, 2021-05-27:
Maybe you have a copy of a working selector app somewhere.

smutek110, 2021-05-27:
I would like to fix it, but my knowledge may not be enough here

smutek110, 2021-05-28:
Maybe you used esp-idf -patched?

smutek110, 2021-05-31:
[image]

smutek110, 2021-05-31:
[image]

smutek110, 2021-06-05:
i fix espboot aplication.
just replace  /espMCUME/espboot/main/component.mk  
with:
#
# Main component makefile.
#
# This Makefile can be left empty. By default, it will take the sources in the
# src/ directory, compile them and link them into lib(subdirectory_name).a
# in the build directory. This behaviour is entirely configurable,
# please read the ESP-IDF documents if you need to do this.
#

COMPONENT_SRCDIRS +=LibFC14

CPPFLAGS += -Wno-error=maybe-uninitialized -Wno-error=sequence-point -Wno-error=implicit-int -Wno-error=pointer-sign -Wno-error=unused-function -Wno-error=implicit-function-declaration -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-error=char-subscripts -Wno-error=attributes -Wno-error=delete-non-virtual-dtor

closed 2021-06-05 by smutek110

smutek110, 2021-06-05:
Now you just need to add the missing emulators keeping in line with your nice gui and psp joy interface. sega megadrive, gb, gbc, sega gamegears, atari7800, snes, lynx, msx1, msx2, pcengine

Jean-MarcHarvengt, 2021-06-06:
happy it works for you!
for me I always get the same error when compiling the espboot

Building partitions from /Users/jean-marcharvengt/Documents/MCUME/MCUME_esp32/espboot/partitions.csv...
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
/Users/jean-marcharvengt/Documents/esp32/esp-idf/components/bootloader_support/src/bootloader_utility.c:54:23: fatal error: esp_efuse.h: No such file or directory
compilation terminated.
make[2]: *** [src/bootloader_utility.o] Error 1
make[1]: *** [component-bootloader_support-build] Error 2
make: *** [/Users/jean-marcharvengt/Documents/MCUME/MCUME_esp32/espboot/build/bootloader/bootloader.bin] Error 2