Jean-MarcHarvengt/MCUME #3: PICOmputer support ?

bobricius, 2021-06-07:
Hi, do you think is possible make support for this board?
[https://hackaday.io/project/179854-picomputer](url)
I can send you complete assembled board.

Jean-MarcHarvengt, 2021-06-08:
Hi Peter,

Nice project, I am surely interested to port to the platform.
Was just wondering how many pins were used for the keyboard matrix?
Minimum I need is for SD card, rest could be remapped to keyboard

I had a similar idea using the Teensy4.0 and 4.1 some time ago. Was not brave enough to solder more than 40 buttons + diodes and I was using an Atmega chip to read matrix over I2C. 
But I like your disposition much more ;-).

Any plan to make suitable for both Teensy4.0/4.1 and Pico?
Both could also support LCD or 8 bits VGA (exclusively)
Challenge would be to route pins to proper location (may be via jumpers?)
Advantage of the Teensy 4.1 is that I could emulate up to the Amiga computer!

If you agree, I will sent you my address. I am from Belgium.
Keep me informed.

Regards,

J-M

bobricius, 2021-06-08:
I wanted make your amiga project :) please write my to bobricius@gmail.com I send you schematic, and design files. then I assembly new board and send to you ;)
Keyboard is matrix 6x6, cursor keys a and shift is on separated group to reduce ghosting.
sd card and displays is on separated spi, there is free some GPIO
also there is standard china display 8pin header, but I prefer if it can work on internal 240x240 st7789 1.54 display