Jean-MarcHarvengt/MCUME #7: MCUME_pico / builds

Kayto, 2022-01-14:
Hi. 

Is it possible to provide precompiled PICO builds?

Compilation for Pico RetroVGA fails as starting message below. 
CMakeLists.txt refers to numerous missing sources from picovga_t4 and missing subdirectory /render.

_CMake Error at CMakeLists.txt:329 (add_executable):
  Cannot find source file:
    picovga_t4/vga_layer.cpp_

alex-taras, 2022-01-16:
Looks like when picovga was stripped down, the CMakeLists file was not cleaned up, just delete everything in the picovga_t4 section that you don't see in the local folder and it'll build.

Jean-MarcHarvengt, 2022-01-16:
I was looking at you problem and indeed I started by cleaning the build directory and no problem.
That means you had some old build directory because I don't push it with the git

Kayto, 2022-01-16:
Thank both for the comments and clarification.
BTW. Great work, thanks.

closed 2022-01-16 by Kayto