Jean-MarcHarvengt/MCUME #8: Write data

pilot-web, 2022-01-17:
Hi,

1) I’ve assembled and installed the teensyuae41. Unfortunately I cannot write any data back to the SD.
The same with the Atari ST.
What do I have to add or modify on sw side?

2) I cannot write any uppercase letters.

I do not have theses issues with the ESP32 VGA hardware and emulated FreeDos or CP/M. Here it is possible to write to SD. But I‘d like to save data to SD on Teensy 4.1, too.

Jean-MarcHarvengt, 2022-01-18:
Floppy drivers have to be adapted for both emu have to be adapted and HDD for UAE. Emuapi has to offer the function to write and that one be used in the drivers. I can have a look this we. Now busy with a new board having keyboard also integrated

pilot-web, 2022-02-06:
Hi Jean-Marc,

I saw that you have published the new keyboard. But USB connected keyboards are more professional and reliable. It is important that they can be mapped to country specific layout. 

Do you now have time to take a look at the write and storage of data to SD?

I do not intend to play any game, but to try and compile in C or Turbo Pascal and save or exchange data via SD. 
Correct USB keyboard / mouse support is required. Like „keyb gr“ setting saved in autoexec.bat for e.g. german layout. 

Even any configuration change should be possible to be saved like in Atari or Amiga emu.

This would be a big step forward.

Regards,
Bernd

Jean-MarcHarvengt, 2022-02-06:
The USB keyboard layout can be selected in the Arduino tools/ "key board layout" for the teensy setup
I compiled executables for Belgian keymap.

I will look for the SD write function this week. I have some plan to improve and add USB keyboard support for all computers.
I also have to merge the autorun feature from the PICO version .

Jean-MarcHarvengt, 2022-02-10:
I added support for writing back to SD on the ST emulator. I just saved the desktop config and it came back at next reboot. The code has been pushed and for the autorun too.

pilot-web, 2022-02-25:
Thank you Jean-Marc.

Are there any plans to enable a write to SD card on Amiga emu, too?

Regards,
Bernd