Jean-MarcHarvengt/MCUME #12: How to switch Atari ST emu to Hires black and white mode

pilot-web, 2022-02-25:
a) How can I switch from the 2 LowRes color modes to permanent HiRes black and white mode?
Do I have to modify a setting in the platform_config.h file for a permanent black and white mode?

b) Please explain die assignment of the 2 user buttons. I found that I could assign another .ST file at the start menu.

Jean-MarcHarvengt, 2022-02-26:
there is currently no way.
On the menu:
USER1 is used to start in hires mode
USER2 to insert second floppy

When running, USER1 to switch between mouse and joystick using the joystick/dpad

Like on a real Atari st, you connect a BW monitor or a Color one.
No way to switch from Color to BW dynamically. When you start with USER1 in HIRES, the VGA quality is less good because of distorsion with the 2 DMAs not synched. There it goes to 640x480, in HIRES/LORES, it uses 640x240.

Still now write routine on the UAE, sorry...

Jean-MarcHarvengt, 2022-02-26:
in fact this is not true, it always uses 640x480, I will see how I can switch between BW and color. But TOS does not allow.
May be via desktop savings?

closed 2022-02-26 by Jean-MarcHarvengt