Jean-MarcHarvengt/MCUME #14: Implementing other emulators

przem360, 2022-03-11:
Hi,
I'm tinkering with MCUME_pico and one old emulator.
My (basic) knowledge of C is not enough to implement existing (old) emulator code in MCUME. 
Author of emulator is using functions like `putpixel()` - in DOS version (I think from Allegro) or `XPutPixel()` - in Linux version to draw a representation of an LCD. Maybe I could get away with using `putpixel()` from `graphics.h` but than I have to send it somehow to MCUME functions (`emu_DrawScreen()` ?)
Could I ask for some help, some good advice on how to go about it?

Jean-MarcHarvengt, 2022-03-11:
I would add the put pixel functions in picoxxx.cpp (all emus have such a file with the main function)
one for 8 and one for 16 bit.

void emu_drawPixel8(int x, int y, unsigned char * color) {
    unsigned char * linept = tft.getLineBuffer(y); 
    linept[x] = color;
}

void emu_drawPixel16(int x, int y, unsigned short * color) {
    unsigned short * linept = tft.getLineBuffer(y); 
    linept[x] = color;
}

and expose them in the emuapi.h

... after this function
extern void * emu_LineBuffer(int line);

extern void emu_drawPixel8(int x, int y, unsigned char * color);
extern void emu_drawPixel16(int x, int y, unsigned short * color);

where you use your putpixel

#include "emuapi.h"

#ifdef USE_VGA
#define RGBVAL8(r,g,b)   ( (((r>>5)&0x07)<<5) | (((g>>5)&0x07)<<2) | (((b>>6)&0x3)<<0) )
emu_drawPixel8(x,y, RGBVAL8(r,g,b))
#else
#define RGBVAL16(r,g,b)  ( (((r>>3)&0x1f)<<11) | (((g>>2)&0x3f)<<5) | (((b>>3)&0x1f)<<0) )
emu_drawPixel16(x,y,RGBVAL16(r,g,b))
#endif

Good luck.

J-M

przem360, 2022-03-12:
Not only you saved my day, you saved my year :relieved:
Big thanks!

closed 2022-03-12 by przem360