Jean-MarcHarvengt/MCUME #20: MCUME_pico : compiling pico64 - getting stuck after VGA display - Needs some help

mchobby, 2022-06-19:
Hi there,
I'm trying to make the pico64 compiling and running. However I'm getting stuck and cannot find how to move forward.
Presently, the VGA displays "C64 Emulator" on the screen and nothing more.

Prepare a sd card (formatted as fat) and extracted the sd.zip to it.
I did also placed the Breakout.PRG into the 'c64 folder'.
I'm using an this 3.3V SD breakout from Adafruit Industries

[3V3 SD Breakout from Adafruit](https://www.adafruit.com/product/4682)
[image]

Here how I did connected it
Breakout --> Pico  / Name_on_Schematic
------------------------------------------------------
3V --> 3V3
GND --> GND
CLK --> GP18 / CLK on schematic
D0/S0 --> GP17 / RX on schematic
CMD/SI --> GP19 / TX on schematic
D3/CS --> GP16 / CS on schematic
D1 --> not connected
Dat2 --> not connected
Det --> not connected

I detected activities on the SD GPIOs (tested with scope).
Did I miss something ?

Thanks for help

Jean-MarcHarvengt, 2022-06-20:
Hi,

Normally it should be fine. 
Try re-formatting the SD card as fat32. You only need the C64 subdirectory with few .prg files inside. 
It should list the files there normally.
if you only see C64 emulator on the VGA output and nothing below it is because it cannot read the SD card.
The important files are /config/iopins.h and /config/platform_config.h. 

Default MCUME_REV2 is selected in platform file !!!!!
//#define MCUME_REV1      1
#define MCUME_REV2      1
so that correspond to
#define SD_SPIREG       spi1
#define SD_SCLK         10
#define SD_MOSI         11
#define SD_MISO         12 
#define SD_CS             13

schematic is MCUME_REV1 so you need to recompile after changing platform file as: 
#define MCUME_REV1      1
//#define MCUME_REV2      1

I hope it helps...

Jean-MarcHarvengt, 2022-06-20:
I personally direct wire the SD slot, I assume the SD breadboard as additional 47K resistor arrays as pull up?
but not sure that would be the issue. AT least recompile for the proper schematic by adapting the platform_config.h file as above.

mchobby, 2022-06-20:
@Jean-MarcHarvengt 
I did indeed compiled with the `#define MCUME_REV1 1` .
Based on your comment, I did check against the content of the `iopins.h` against the wiring I made. There was an error. So here is **the correct wiring for the SD on MCUME_REV1**. 

3V --> 3V3
GND --> GND
CLK --> GP18 / CLK on schematic
D0/S0 --> **GP16**
CMD/SI --> GP19 / TX on schematic
D3/CS --> **GP17** 
D1 --> not connected
Dat2 --> not connected
Det --> not connected

And now, it works and show me the file.

mchobby, 2022-06-20:
@Jean-MarcHarvengt ,

You'd recommend the to use the MCUME_REV2.
Could you explain me why MCUME_REV2 would be a better choice? (I'm not comfortable with MCUME yet).

Cheers,
Dominique

Jean-MarcHarvengt, 2022-06-21:
HI,

You are indeed right, RX and CS have to be swapped on the REV1 schematics.
REV1 was my initial try out.
In fact REV2 started because I had to port MCUME to below platform from Peter Misenko.

https://github.com/bobricius/PICOmputer

At a certain point we had a version with VGA and LCD display at same time so we had to switch the SD card to another SPI so Rev2 was made for me to test without the PICOMPUTER in my hands at the beginning.
REV2 pinout is just more compatible with pinout of PICOMPUTER and PICORETROVGA of Peter.
Just a lot of history...

Binaries in /bin are available for all the platforms but I have to say, the REV1 was just pure wiring on a breadboard.
for REV2 I had at least made a small hand made PCB.
Forgive me if you find some mistakes.
PICOMPUTERMAX is in fact the more spread and tested version.
You will also find a lot of incompatibilities in the emulators but it is just funny to see all these running on a 5Euros platform.
All my PICOs comes from your shop so....it is pretested for you already ;-)

closed 2022-06-21 by Jean-MarcHarvengt