Jean-MarcHarvengt/MCUME #22: Add sound to teensydoom for Teensy 4.1

xavier37, 2022-08-27:
Hi Jean Marc,

Would it be possible to add I2S sound support on Teensydoom ? 
I have found videos where sound is working on ESP32 board with less memory than T4.1 and I2S capability, figuring out why it does not work with this platform 
Thanks

Jean-MarcHarvengt, 2025-08-25:
> Hi Xavier37 - Hi Jean-Marc
> Well... It's been almost 3 years ago and it seems the Question is still
> open, althought PT8211 is meantime Part of the Setup.
> I have Problems too to get the PT8211 running (for testing with DOOM), no
> Sound. I do not understand why it isn't clearly described where to find the
> settings for I²S Audio and is it anyhow "connected" to "HAS_SND" oder
> "PT8211". Strange at all... I've wrote something in PJRC Forum under
> Jean-Marc's Topic "Introducing the T-COMPUTER platform". To me it is
> unpredictable to what Marc are doing. It seems he'd gived up to keep his
> Branch alive. Is you able (meantime) to understand the sound configuration
> and how it works? If yes, it would be a nice move of you to share this
> knowledge with me.
> Best Regards,
> The Doctor
> (blooth)
>
>
> https://forum.pjrc.com/index.php?threads/introducing-the-t-computer-platform.70517/#post-361149

Sorry I have been focussing on other projects the past 3 years. I never
tried to have sound in doom. Most emulators in mcume have sound in the
teensy 41 branch  you really cannot find inspiration there?

Blooth, 2025-08-25:
Good Morning Jean-Marc - You're alive! Woohoo
At first: It's amazing what you and the other guys have created. The Dream of every "small kid in us" to have all the Machines combined in a little one. So to say: "We're entering the Micro Cosm".

Don't mind, Jean-Marc, I know that other Projects maybe gets quicker in focus and the actual one slides slowly more and more into the background.

Yes, in a certain way I see a lot things what's "to do" and there's actually no lack of inspirations to me or anyone else. But my coding knowledge is still and heavily framed. And I suppose that a lot of People do not dare to try it because of it's "tonns" of Libs and cross linkings. I suspect 'gensnd' and 'snd' itself which seems not to be routed to "PC SPEAKER" or SoundBlaster emulated IRQ/DMA to grab the Audio data.

Oh my gosh, I saw right now that Frank was already on Bord with you and discussing the DMA/IRQ Problems with VGA Libs.
But why he didn't linked the PT8211 immediately to DOOM (or why hasn't PT8211 not it's own "Preferences File" to link with the DOOM Library? I suppose it was an "unwanted accident" by discussing the MCUME and nobody hast thought furtermore to link Doom. That can happen of course. So, I can see 2 People who's able to make it possible: You and Frank of course. Advice and hints are welcome too to get deeper into it. I would help as good as I can.

As good as I can remember, there was a config file for doom to select the Sound Setup, f.g. GUS or Blaster. Isn't possible to make a "couple" more Includes to have the Sound settled with PT8211?

If I didn't metioned it, I saw in events.h and audio.c some lines, which might explain what be would the reason for non-audio in DOOM. Perhaps you can confirm my thoughts.

>#include "shared.h"

#include "memory.h"
#include "custom.h"
#include "gensound.h"
#include "sounddep/sound.h"
#include "events.h"
#include "audio.h" <

Many links to manage and I think that's the key to everything else.

BTW (TeensyUAE): The Sound accuracy doesn't match the original speed of 68k CPU, is it possible to decrease the FPS Rate to match the origial Audio speed?

Blooth, 2025-11-06:
Hi Xavier,
as I mentioned above, I did some Code in several .h Files to grab the Sound direct from Doom-Code. It's 4-Channel true Stereo Output and it works flawless. I've made Jean-Marc an offer, but he seem no longer interested in his own Project. Unfortunately I'm not able to fork this here... Maybe I can make a Branch for this...

Jean-MarcHarvengt, 2025-11-15:
Hi Blooth,

Can you provide the changes that you did to me that I can merge it in the main branch?
Thanks!

Blooth, 2025-11-16:
> Hi Blooth,
> 
> Can you provide the changes that you did to me that I can merge it in the main branch? Thanks!

I'm so sorry for the waiting. I have right now a Dead in the Family (my Brother has passed away by a gun shot himself) and it need a couple days to handle this. I'll return asap. Notice: I made it for Teensy4.1 but I'm on it to make it for ESP32 too... Can you leave me any Contact of you to stay in Touch?

Jean-MarcHarvengt, 2025-11-16:
Hi Blooth,
Not an easy time, ...sorry about that...
You can send me a mail to my github email when you want.

Regards,

J-M