Jean-MarcHarvengt/MCUME #27: Pico Display

YnotZer0, 2022-12-25:
I've successfully setup, built & created the mcume.uf2 file - so I know my setup/config is now working good.
All I am attempting to do is run the Atari800 emulator on a Pico with a waveshare screen+sd-card attached.  The specs says it is using the ST7789 driver - so should be good.
However, when I drop the .uf2 on and the Pico reboots, the backlight comes on, but I do not get any output on the screen.
It is this screen here: https://www.waveshare.com/wiki/Pico-ResTouch-LCD-2.8
Is there anyway to debug what is happening?  btw - I only have the screen attached (that has a valid FAT sd-card with the folders & files created), I do not have anything else attached though - would that prevent anything showing on the screen?

YnotZer0, 2022-12-30:
ah, okay - found the TFT_* constant setting values within iopins.h - that's where I need to make changes for the CS, DS, MOSI etc... values.  I'll have a dig around and see what I can do.
I'd still like to know, if possible, how to do debugging of the PICO when it is executing the emuapi.cpp code.  I see a lot of printf() commands, but don't understand how I can visualise them.  If you could add any extra info about debugging to the readme, that would be very helpful.

closed 2022-12-30 by YnotZer0

Jean-MarcHarvengt, 2022-12-30:
happy you could find it!

in CMakeLists.txt you can enable one of the macro below to have logs
#pico_enable_stdio_uart(mcume 0)
#pico_enable_stdio_usb(mcume 0)

same file where you can find the various emu target that you can comment/uncomment