Jean-MarcHarvengt/MCUME #28: Out of Memory with picospeccy

saarbastler, 2022-12-30:
Hi all,
great project !

I have an issue with PIComuter Max and picospeccy. After starting a game it just stops working, only a Reset takes it back to the main menu.  The MCUME\MCUME_pico\bin\PICOMPUTERMAX\mcume_picospeccy.uf2 is working fine, but I want to make code modifications.

So I added 
pico_enable_stdio_usb(mcume 1)
target_compile_definitions(mcume PRIVATE PICO_DEBUG_MALLOC=1)

to the CMakeLists.txt and I got this output:
`sd_spi_go_low_frequency: Actual frequency: 100000
V2-Version Card
R3/R7: 0x1aa
R3/R7: 0xff8000
R3/R7: 0xc0ff8000
Card Initialized: High Capacity Card
SD card initialized
SDHC/SDXC Card: hc_c_size: 15199
Sectors: 15564800
Capacity:     7600 MB
sd_spi_go_high_frequency: Actual frequency: 1000000
malloc 1120 20034430->20034890
malloc 1120 20034430->20034890
malloc 1120 20034430->20034890
malloc 1120 20034430->20034890
malloc 1120 20034430->20034890
malloc 1120 20034430->20034890
malloc 1120 20034430->20034890
malloc 1120 20034430->20034890
SD initialized, files found:
6
malloc 1120 20034430->20034890
new filepath is
spec/SpaceIntruders.z80
malloc 1120 20034430->20034890
malloc 320 20034430->20034570
could allocate dynamic
320
malloc 49152 00000000->0000C000

*** PANIC ***

Out of memory`

malloc did not work with 49152 bytes, not enough memory left. Do you have an idea, why your uf2 file is working, but my local build does not ?

Regards,
saarbastler

Jean-MarcHarvengt, 2022-12-30:
The file is probably too big? have you tried a smaller z80 file?
As I recall only spectrum 48k is supported, not 128k games

saarbastler, 2022-12-31:
I'm interested in Spectrum 48k only, because I had one, long time ago ...

The system allways allocate 48k Ram, independent of the file size: [spec.c](https://github.com/Jean-MarcHarvengt/MCUME/blob/bb3ec50b4bd9d6e0f90a3e3114ce50da5d0f0540/MCUME_pico/picospeccy/spec.c#L253)

All the z80 files I tested, are working with your build [mcume_picospeccy.uf2](https://github.com/Jean-MarcHarvengt/MCUME/blob/master/MCUME_pico/bin/PICOMPUTERMAX/mcume_picospeccy.uf2), so I guess it's just a cmake configuration issue.

saarbastler, 2023-01-01:
Hi,

I made a fork and modified the code not to use malloc reading z80 files: https://github.com/saarbastler/MCUME/commit/d949173cbc4cf4b53df6265b93053fd06faa2830 

It's working fine, but I was not able to test the "New Version" part: https://github.com/saarbastler/MCUME/blob/d949173cbc4cf4b53df6265b93053fd06faa2830/MCUME_pico/picospeccy/zx_filetyp_z80.c#L234
because I didn't found a "new version" z80 file