Jean-MarcHarvengt/MCUME #33: How to set teensy4.1 nofrendo to PAL/50hz, emulator and vga?

Raduq91, 2023-06-17:
How to set teensy4.1 nofrendo to PAL/50hz, emulator and vga?

Jean-MarcHarvengt, 2023-06-19:
VGA is only 60hz, no emulator really supports pal. Vsync is defined in ino file. It normally vsyncs on VGA but can be overruled to sync on timer instead. Timer often set to 16666 ms (NTSC)

Raduq91, 2023-06-19:
> VGA is only 60hz, no emulator really supports pal. Vsync is defined in ino file. It normally vsyncs on VGA but can be overruled to sync on timer instead. Timer often set to 16666 ms (NTSC)

So these don't do anything?
teensynofrendo/nes.c

```
#define  NES_CLOCK_DIVIDER    12
//#define  NES_MASTER_CLOCK     21477272.727272727272
#define  NES_MASTER_CLOCK     (236250000 / 11)
#define  NES_SCANLINE_CYCLES  (1364.0 / NES_CLOCK_DIVIDER)
#define  NES_FIQ_PERIOD       (NES_MASTER_CLOCK / NES_CLOCK_DIVIDER / 60)

#define  NES_RAMSIZE          0x800

#define  NES_SKIP_LIMIT       (NES_REFRESH_RATE / 5)   /* 12 or 10, depending on PAL/NTSC */
```

teensynofrendo/nes.h

```
/* NTSC = 60Hz, PAL = 50Hz */
#ifdef PAL
#define  NES_REFRESH_RATE     50
#else /* !PAL */
#define  NES_REFRESH_RATE     60
#endif /* !PAL */

#define  MAX_MEM_HANDLERS     32
```

I tried to change settings, I thought I did it wrong.

Tried to change settings in     VGA_t4/VGA_t4.cpp
```
#define frame_freq     60.0     // Hz
#define line_freq      31.46875 // KHz
#define pix_freq       (line_freq*800) // KHz (25.175 MHz)

// pix_period = 39.7ns
// H-PULSE is 3.8133us = 3813.3ns => 96 pixels (see above for the rest)
#define frontporch_pix  20 //16
#define backporch_pix   44 //48
```
Nothing worked.

My goal was to set the emulator and vga to 50hz, and try to set vga_t4 to double VT(vertical total).
On PC double VT decreases input latency with vsync on. And I have a benq xl2411 that has backlight strobing, and having larger VT reduces image crosstalk.
https://dough.community/t/blur-buster-eve-spectrum-strobe-tuning-completion/29085/35

JustMLC4real, 2023-10-05:
Did you manage to figure it out? I'm kind of struggling with a similar problem and it does indeed seem that the function PAL actually never gets defined in Nofrendo at all...